JetsonHacksNano/CSI-Camera #1: Camera module V1.3 on Jetson?

maximecb, 2019-04-09:
Hi there!

I'm trying to get a camera module V1.3 to work with a Jetson Nano. Unfortunately, it's not showing up as `/dev/video*`. I'm wondering if there are some packages I would need to install first, or if only the camera module V2 is supported on the Jetson Nano. I'm using the stock Nano ubuntu image, not the one created for the Jetbot.

Any help/pointers much appreciated.

liurongfang, 2019-04-10:
> 嗨，您好！
> 
> 我正在尝试使用相机模块V1.3与Jetson Nano配合使用。不幸的是，它没有出现`/dev/video*`。我想知道是否有一些我需要先安装的软件包，或者如果Jetson Nano只支持相机模块V2。我正在使用Nano ubuntu图像，而不是为Jetbot创建的图像。
> 
> 谢谢你的精彩内容。任何帮助/指针非常感谢。

hello,maximecb,
rpi v1 camera is not be supported,so sad,I am using it,,,this is offical answer,https://devtalk.nvidia.com/default/topic/1049605/jetson-nano/-raspberry-pi-version-1-camera-does-not-work/,
maybe I should use Logitech C920 or RPI V2 IMX219 cameras.

maximecb, 2019-04-10:
If you're looking for a cheap webcam to use with the Jetson, I found this $4 ebay webcam works out of the box: https://www.ebay.ca/itm/Mini-USB-2-0-PC-Camera-HD-Webcam-Camera-Web-Cam-For-Laptop-Desktops-E-CA/312515707441?hash=item48c3634231&enc=AQADAAAC8FjVrDbVsZ8oH%2F8PNHtt9VX4%2Fw7FZcmMuqsX8uaFEduVkWmkUeIV6VdVe9NdEVObrKxguCZ%2F8SgTzUhciyoxjyWy1UJsbF23pFXkJIYiQNKZgImjzTQrbjO%2FzuPLu%2B32IHkwjCgBKWt%2FghBusSIbs0dS%2F484wd04I%2B1recC4pqlTswACkCyHLyo1vAAWibDkvWBZBk9ozSaz7Q83xkqrXJQjRMizE7faGaGEnuLugJ90%2F6TVsz%2FYXmMQcwxaay7OBJpNnJ36XscUYxXHaWjKyXX2vC%2FmbLSo2n3fllamSdrcnMEKDwdEFN%2F8fv11ijn7ARxoSlQJJ1%2BGePDjtII1p2gCRu6pth%2BvFenpYX1QyJwHW1Dn9ZE2%2B5oijxMgLrkjvda3ITrXWOVz1hTtA%2BmCX%2B1gAFZXEXzT1jfmSfOdDmkBJ56cqsxVCfMjwb%2FUVcIF3O6kcXiJ6Plwm50qx689UJe%2BZeOYlVInqMIS5nE6AouFqlbkRUFE67k2zOqbLJJ49FL9SYQeCdlZybTM5BoA9E%2FOhNmfYsxFEwPyoONzVAO7FzjXrwnI5Hw72sY%2F17LX5qBZ3g3yi3CFjNUiUbvLmI9%2B3AEnmHw7RBpA6lT3uNyXKlzmvMvcxGgRrs6g7jCuUB4KTHJtgViXJJ8%2F3p4ZQK4guvKE7YO8dM94uEcEZdYO5ZJizgHFXD14jybIck80uHyVJtTRwQRbGAupVU%2F218cqZneCPgqShlcRqRRWAcGxTQ9OTisAwb2n7y6%2BulJ%2BaQB9EAvrnTZAox%2FRv%2F0Vj1gj2D1fZKuTRijsuL2Dvcy84MPpZ%2BjQUOLY%2BFcmQRbryMxrBOC4TPQOVuJvGdNwXhK%2BUqopROnjJjwtzW7N5Wb9Cgzrtx5LbKC9BUbI6uQVddYkv3TxNgx4xxka%2BksaG%2FRurIKyFGKkOSQyaK4jRAVOABaWDU14vE9szGKu1kUoiFyWia8dBQGijf1CF1LAjv%2FUSPi8zdJmDYKbL8U5d7wB&checksum=312515707441fc1c67f8ac91470e95df30b11cbc686f

liurongfang, 2019-04-10:
Yes,I am looking for,ebay can't ship to my location,I found this camera in [$2.5 taobao](https://item.taobao.com/item.htm?spm=a1z0d.6639537.1997196601.70.69577484ktlapT&id=586975488691)，I will buy and test it ,thank you very much!

JetsonHacksNano, 2019-11-10:
The repository is for installing a RPi V2 camera. NVIDIA does not support the V1 cameras.

closed 2019-11-10 by JetsonHacksNano