JetsonHacksNano/CSI-Camera #2: Changes to examples for USB Webcam (Logitech)

DennisFaucher, 2019-04-17:
Hello. Thank you for the examples. What changes need to be made to use a USB webcam on the Nano? Your gst-launch-1.0 command results in these errors:
Setting pipeline to PAUSED ...

Using winsys: x11 
Pipeline is live and does not need PREROLL ...
Got context from element 'eglglessink0': gst.egl.EGLDisplay=context, display=(GstEGLDisplay)NULL;
Setting pipeline to PLAYING ...
New clock: GstSystemClock
GST_ARGUS: Creating output stream
CONSUMER: Waiting until producer is connected...
GST_ARGUS: Available Sensor modes :
Caught SIGSEGV
#0  0x0000007f81b64048 in __GI___poll (fds=0x55a0af67a0, nfds=547638157744, timeout=<optimized out>) at ../sysdeps/unix/sysv/linux/poll.c:41
#1  0x0000007f81c71d80 in  () at /usr/lib/aarch64-linux-gnu/libglib-2.0.so.0
#2  0x00000055a0855af0 in  ()
Spinning.  Please run 'gdb gst-launch-1.0 6380' to continue debugging, Ctrl-C to quit, or Ctrl-\ to dump core.
(Argus) Error Timeout:  (propagating from src/rpc/socket/client/ClientSocketManager.cpp, function send(), line 137)
(Argus) Error Timeout:  (propagating from src/rpc/socket/client/SocketClientDispatch.cpp, function dispatch(), line 87)
Error generated. /dvs/git/dirty/git-master_linux/multimedia/nvgstreamer/gst-nvarguscamera/gstnvarguscamerasrc.cpp, threadExecute:243 Stream failed to connect.
Error generated. /dvs/git/dirty/git-master_linux/multimedia/nvgstreamer/gst-nvarguscamera/gstnvarguscamerasrc.cpp, threadFunction:177 (propagating)

JetsonHacksNano, 2019-11-10:
This repository is for a RPi v2 camera. 
Typically people would set the gst-launch to the webcam device:
v4l2src device="/dev/video0"

closed 2019-11-10 by JetsonHacksNano

JetsonHacksNano, 2019-11-10:
You are welcome. Unfortunately it took me a long time to get around to this.