JetsonHacksNano/CSI-Camera #3: Unable to open camera with Python3

Prologueyan, 2019-05-24:
It works well use python2 ，with log ：
nvbuf_utils: Could not get EGL display connection
And  unable to open camera use python3

plcpinho, 2019-05-25:
Same error when running "`python simple_camera.py`":
"`Unable to open camera`"

hvacking, 2019-06-24:
Using winsys: x11
Pipeline is live and does not need PREROLL ...
Got context from element 'eglglessink0': gst.egl.EGLDisplay=context, display=(Gs
Setting pipeline to PLAYING ...
New clock: GstSystemClock
**
ERROR:/dvs/git/dirty/git-master_linux/multimedia/nvgstreamer/gst-nvegltransform/ed: (gst_query_get_n_allocation_pools (query) > 0)

aydal, 2019-11-08:
Could not run simple_camera.py with both python2 and python3. Got `unable to open camera` each time. Is this resolved?
```gst-launch-1.0 nvarguscamerasrc ! 'video/x-raw(memory:NVMM),width=3820, height=2464, framerate=21/1, format=NV12' ! nvvidconv flip-method=0 ! 'video/x-raw,width=960, height=616' ! nvvidconv ! nvegltransform ! nveglglessink -e``` works fine.
Tried to change parameters in python script to same as gstream command but could not access camera

aydal, 2019-11-09:
Resolved. OpenCV wasn't build with Gstreamer.
use `print(cv2.getBuildInformation())` in python2 and\or python3 to confirm. It worked after I purged opencv completely and build using this link.
https://jkjung-avt.github.io/opencv-on-nano/
For some reason Jetson hacks script to build opencv didnt work for gstream

JetsonHacksNano, 2019-11-10:
The default installation for a Jetson Nano is OpenCV 3.3.1. This is GStreamer compatible. The Python issue has been resolved in newer releases of the repository.

closed 2019-11-10 by JetsonHacksNano

biel-jony, 2021-07-20:
I got 'nvbuf_utils: Could not get EGL display connection' problem too, 
when I connect the monitor with HDMI , it worked.