JetsonHacksNano/CSI-Camera #5: How to send the video of the raspberry pi camera module with Jetson Nano to any server.

ashishbharadwaj100, 2019-07-08:
How to send the video of the raspberry pi camera module with Jetson Nano to any server. Does anyone have done this??

nijisakai, 2019-07-16:
try *mgpg-streamer* as used on RPI

ashishbharadwaj100, 2019-07-17:
Yes I tried nijisakai 
they have given this link also 
https://devtalk.nvidia.com/default/topic/1055213/jetson-nano/mjpg-streamer-/

Then I disable the opencv with this configuration and given this command.

cd mjpg-streamer-experimental

From this link I got the idea of disable the open cv
https://devtalk.nvidia.com/default/topic/1055213/jetson-nano/mjpg-streamer-/

cmake -DPLUGIN_INPUT_OPENCV=OFF
make
sudo make install
And 
Apt-get install libv4l-dev

And Customize my linux command to 
./mjpg_streamer -o "output_http.so -w ./www" -i "
-launch-1.0 nvarguscamerasrc ! 'video/x-raw(memory:NVMM),width=3820, height=2464, framerate=21/1, format=NV12' ! nvvidconv flip-method=0 ! 'video/x-raw,width=960, height=616' ! nvvidconv ! nvegltransform ! nveglglessink -e"
[image]

JetsonHacksNano, 2019-11-10:
No comment in over 90 days. Closing.

closed 2019-11-10 by JetsonHacksNano