JetsonHacksNano/CSI-Camera #15: No cameras available

FXRer, 2019-10-15:
Hi,
I'm using the [MIPI-CSI2 camera from e-Con Systems](https://www.e-consystems.com/nvidia-cameras/jetson-nano-cameras/3mp-mipi-camera.asp) and got following error :  
`$ python3 simple_camera.py 
nvarguscamerasrc ! video/x-raw(memory:NVMM), width=(int)1280, height=(int)720, format=(string)NV12, framerate=(fraction)60/1 ! nvvidconv flip-method=0 ! video/x-raw, width=(int)1280, height=(int)720, format=(string)BGRx ! videoconvert ! video/x-raw, format=(string)BGR ! appsink
Error generated. /dvs/git/dirty/git-master_linux/multimedia/nvgstreamer/gst-nvarguscamera/gstnvarguscamerasrc.cpp, execute:521 No cameras available`

Even-though the camera is available as `/dev/video0` and can show the video.

Could you please help me to point out the problem please?

Thanks in advance,
Pascal K.

kannan-m, 2019-10-21:
nvarguscamerasrc will detect cameras that make use of NVIDIA ISP in Jetson NANO. e-con's e-CAM30_CUNANO has its own on-board ISP, so it will not be detected by nvarguscamerasrc. 
You can use v4l2src instead of nvarguscamerasrc to get frames in UYVY format. 

e.g. 
`$ gst-launch-1.0 v4l2src ! "video/x-raw, format=(string)UYVY, width=(int)2304, height=(int)1536" ! nvvidconv ! "video/x-raw(memory:NVMM), format(string)I420, width=(int)1920, height=(int)1080" ! nvoverlaysink overlay-w=1920 overlay-h=1080 sync=false`

Please find more examples on how to use gstreamer in the "Gstreamer Usage Guide" document provided with the Release package.

JetsonHacksNano, 2019-11-10:
Thank you for the explanation @kannan-m 
Because this repository is specifically for the RPi version 2.0 camera, I'll close this here.

closed 2019-11-10 by JetsonHacksNano

sanariaz154, 2020-05-13:
I am having a similar issue. I am able to run `gst-launch-1.0 nvarguscamerasrc ! nvoverlaysink
` on host machine succefully but when i run it inside a docker  container , i get "No camera availble " error. i am able to see /dev/video0 inside docker container. 
Any suggestion would be helpful

mattedminster, 2020-05-18:
@sanariaz154  Did you end up figuring this out. I am facing the same challenge

sanariaz154, 2020-06-01:
@mattedminster I was facing this issue because of incompatible drivers version inside docker. Those were the driver that nvarguscamerasrc uses to talk to Camera. Instead of installing the driver explicitly inside docker, I used nvidia docker mount plugin which automatically mounted the required host drivers in docker and the issue was solved

mhaboali, 2020-07-07:
Hi @sanariaz154 could you explain how did you use with your existing docker image?

I have one docker image where the software runs on Jetson Nano but it needs the camera stream and simple_camera.py cannot access the camera. And when I tried to access /dev/video0, OpenCV cannot open the video either.

Could you help with this, please?