JetsonHacksNano/CSI-Camera #21: [BUG] face_detect.py crash on detectMultiScale

EnricoBeltramo, 2019-12-31:
**Describe the issue**
starting face_detect.py, I have follow error:
Traceback (most recent call last):
  File "/home/jetson/CSI-Camera/face_detect.py", line 83, in <module>
    face_detect()
  File "/home/jetson/CSI-Camera/face_detect.py", line 58, in face_detect
    faces = face_cascade.detectMultiScale(gray, 1.3, 5)
cv2.error: OpenCV(3.4.6) /home/jetson/src/opencv-3.4.6/modules/objdetect/src/cascadedetect.cpp:1698: error: (-215:Assertion failed) !empty() in function 'detectMultiScale'

**What version of L4T/JetPack**
L4T/JetPack version:
nv-jetson-nano-sd-card-image-r32.3.1

**What version of OpenCV**
OpenCV version:
3.4.6

**Python Version**
Python version if applicable:
3.6

**To Reproduce**
Steps to reproduce the behavior:
For example, what command line did you run?
run face_detect.py

**Expected behavior**
A clear and concise description of what you expected to happen.
program don't crash

**Additional context**
Add any other context about the problem here.

JetsonHacksNano, 2020-01-01:
Please try the latest release.

EnricoBeltramo, 2020-01-01:
I found the issue. The xml files for detection:
        "/usr/share/opencv4/haarcascades/haarcascade_frontalface_default.xml"
        "/usr/share/opencv4/haarcascades/haarcascade_eye.xml"
wasn't present (or may be in diefferent position) in my opencv installation. I downloaded the from opencv source repository and changed the path. Now it works fine

closed 2020-01-01 by EnricoBeltramo