JetsonHacksNano/CSI-Camera #22: [BUG]

lackdaz, 2020-02-17:
**Describe the issue**
```
  File "face_detect.py", line 83, in <module>
    face_detect()
  File "face_detect.py", line 58, in face_detect
    faces = face_cascade.detectMultiScale(gray, 1.3, 5)
cv2.error: OpenCV(4.2.0) /tmp/build_opencv/opencv/modules/objdetect/src/cascadedetect.cpp:1689: error: (-215:Assertion failed) !empty() in function 'detectMultiScale'```

I've seen an issue like this in #21 , I've checked his fix but both files were present in my opencv build.

**What version of L4T/JetPack**
L4T/JetPack version: Linux 4.9.140-tegra aarch64

**What version of OpenCV**
OpenCV version: 4.2.0
I used mdegans's build script

**Python Version**
Python version if applicable: Python 2.7 (cv4 didn't seem to be installed for Python3, still defaulting to 3.3.1)

**To Reproduce**
Steps to reproduce the behavior:
python face_detect.py

**Expected behavior**
Don't crash! 

**Additional context**
Should I be running your OpenCV4 build script instead? Is it too late now?

By the way, I would like to take the opportunity to express my gratitude for your video content. It's pull me out of one-too-many pipeline problems. Its such terrific content that deserves way more attention

lackdaz, 2020-02-17:
I managed to fix this by correcting the filepath. mdegans's build script installs it as `/usr/share/OpenCV` vs `/usr/share/opencv4` for yours.

Ok now that I have it working, but sadly, only the video comes on and I'm missing the blue lines and green boxes :(

closed 2020-02-17 by lackdaz

Gralios, 2020-05-11:
what i need make to complete the face identification?