JetsonHacksNano/CSI-Camera #26: Compatible with new Raspberry Pi Camera?

sebo361, 2020-06-15:
Hello, 

i have the Jetson Nano B01 Kit and the [new Raspberry Pi Camera ](https://www.raspberrypi.org/products/raspberry-pi-high-quality-camera/). I connected the camera in slot CAM0 with the pins towards the Jetson main board. However, the Jetson does not find the camera - there is no `/dev/video`.

Do I need to change or upgrade anything in order to make this new Raspberry Pi Camera work? Any camera driver missing?

Thanks and best,
Sebastian

arushk1, 2020-07-05:
The new Cam V2.1 works well with B01. Please check your camera and/or connection again.
Alternatively. Restart your device with cam plugged in.

sebo361, 2020-07-05:
Hi @arushk1, i was curious about connecting the new HD camera: [https://www.raspberrypi.org/products/raspberry-pi-high-quality-camera/](https://www.raspberrypi.org/products/raspberry-pi-high-quality-camera/). A team is working on the camera driver: [https://www.raspberrypi.org/products/raspberry-pi-high-quality-camera/](https://www.raspberrypi.org/products/raspberry-pi-high-quality-camera/)

closed 2020-07-05 by sebo361