JetsonHacksNano/CSI-Camera #27: Unable to open camera

hbzhang, 2020-07-10:
when I run this, 

python simple_camera.py 

I got:

nvarguscamerasrc ! video/x-raw(memory:NVMM), width=(int)1280, height=(int)720, format=(string)NV12, framerate=(fraction)60/1 ! nvvidconv flip-method=0 ! video/x-raw, width=(int)1280, height=(int)720, format=(string)BGRx ! videoconvert ! video/x-raw, format=(string)BGR ! appsink
Unable to open camera

Why? Thanks

BaptisteAmato, 2020-08-18:
Hi,

Which version of OpenCV are you using? I had the same problem while using cv2 3.2, it worked after upgrading it.

lxc0531, 2022-02-13:
> 

when I install opencv_contribute_python,it dosen't work

sparkster98, 2022-02-20:
These two lines got it working for me:

`sudo apt-get install libavcodec-dev libavformat-dev libswscale-dev
sudo apt-get install libgstreamer-plugins-base1.0-dev libgstreamer1.0-dev`

found here:

[https://docs.opencv.org/3.4/d2/de6/tutorial_py_setup_in_ubuntu.html](url)

Installing opencv from source will likely help but test after running those two lines above first.

jksojitraa, 2022-03-01:
hello @BaptisteAmato ,
 
which version did you use
to install

thanks in advance