JetsonHacksNano/CSI-Camera #31: No cameras available

reshesh, 2020-10-14:
gst-launch-1.0 nvarguscamerasrc sensor_id=1 ! nvoverlaysink
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
Error generated. /dvs/git/dirty/git-master_linux/multimedia/nvgstreamer/gst-nvarguscamera/gstnvarguscamerasrc.cpp, execute:557 No cameras available
Got EOS from element "pipeline0".
Execution ended after 0:00:00.062571012
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...

mhaboali, 2020-11-23:
Any updates on this issue, I have the same here on Jetson Xavier Nx

micuentadecasa, 2022-11-03:
thanks for the info @caesarakalaeii

ehdrb92, 2025-03-27:
I had the same problem. In my case, it was a flat cable problem connecting the CSI camera module. I replaced the cable with a new one and it worked fine.

I checked the system log with the "dmesg" command and found the log "Imx219_board_setup: error during i2c read probe" to find the related issue.

I found out about the possibility of a hardware problem from the following link and fixed the problem by replacing the cable.

https://forums.developer.nvidia.com/t/imx219-board-setup-error-during-i2c-read-probe/243759