JetsonHacksNano/CSI-Camera #32: face_detect.py : Jetson reboots while executing

cristinaluna, 2020-11-13:
Hi!

I'm trying to execute face_detect.py on my jetson nano 2Gb (jp441)

It freezes whithout prompting the camera screen, in the console it says `CONSUMER: Producer has connected; continuing` and then reboots the platform

Thank you!

Gyansai, 2021-01-02:
me too facing the same error 
```
CONSUMER: Waiting until producer is connected...
GST_ARGUS: Available Sensor modes :
GST_ARGUS: 3264 x 2464 FR = 21.000000 fps Duration = 47619048 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 3264 x 1848 FR = 28.000001 fps Duration = 35714284 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1920 x 1080 FR = 29.999999 fps Duration = 33333334 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1640 x 1232 FR = 29.999999 fps Duration = 33333334 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1280 x 720 FR = 59.999999 fps Duration = 16666667 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1280 x 720 FR = 120.000005 fps Duration = 8333333 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: Running with following settings:
   Camera index = 0 
   Camera mode  = 3 
   Output Stream W = 1640 H = 1232 
   seconds to Run    = 0 
   Frame Rate = 29.999999 
GST_ARGUS: Setup Complete, Starting captures for 0 seconds
GST_ARGUS: Starting repeat capture requests.
CONSUMER: Producer has connected; continuing.

```
jetpack version 4.4, camera - IMX219-77IR , OpenCV - 4.1.1
urgency -very urgent

cristinaluna, 2022-09-20:
I'm going to close this issue due to inactivity. 

It could be a problem with the camera! but needs to be investigated

closed 2022-09-20 by cristinaluna