JetsonHacksNano/CSI-Camera #36: [BUG] can't start the camera

KevorkSulahian, 2021-02-06:
**Describe the issue**
I simply can't start the camera

**What version of L4T/JetPack**
L4T/JetPack version:
nvidia-l4t-core	32.5.0-20210115145440

**What version of OpenCV**
OpenCV version: 4.1.1

**Python Version**
Python version if applicable: 3.6.9

**To Reproduce**
Steps to reproduce the behavior:
run the sample .py code

**Expected behavior**
A clear and concise description of what you expected to happen.
run the code ad open the camera

**Additional context**
Add any other context about the problem here.
## this is the error

nvarguscamerasrc ! video/x-raw(memory:NVMM), width=(int)1280, height=(int)720, format=(string)NV12, framerate=(fraction)60/1 ! nvvidconv flip-method=0 ! video/x-raw, width=(int)1280, height=(int)720, format=(string)BGRx ! videoconvert ! video/x-raw, format=(string)BGR ! appsink
GST_ARGUS: Creating output stream
CONSUMER: Waiting until producer is connected...
GST_ARGUS: Available Sensor modes :
GST_ARGUS: 3264 x 2464 FR = 21.000000 fps Duration = 47619048 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 3264 x 1848 FR = 28.000001 fps Duration = 35714284 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1920 x 1080 FR = 29.999999 fps Duration = 33333334 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1640 x 1232 FR = 29.999999 fps Duration = 33333334 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1280 x 720 FR = 59.999999 fps Duration = 16666667 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1280 x 720 FR = 120.000005 fps Duration = 8333333 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: Running with following settings:
   Camera index = 0 
   Camera mode  = 5 
   Output Stream W = 1280 H = 720 
   seconds to Run    = 0 
   Frame Rate = 120.000005 
GST_ARGUS: Setup Complete, Starting captures for 0 seconds
GST_ARGUS: Starting repeat capture requests.
CONSUMER: Producer has connected; continuing.
nvbuf_utils: dmabuf_fd -1 mapped entry NOT found
nvbuf_utils: Can not get HW buffer from FD... Exiting...
[ WARN:0] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (1757) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module nvarguscamerasrc0 reported: CANCELLED

(python3:9699): GStreamer-CRITICAL **: 22:47:20.231: gst_mini_object_set_qdata: assertion 'object != NULL' failed
(Argus) Error EndOfFile: Unexpected error in reading socket (in src/rpc/socket/client/ClientSocketManager.cpp, function recvThreadCore(), line 266)
(Argus) Error EndOfFile: Receive worker failure, notifying 1 waiting threads (in src/rpc/socket/client/ClientSocketManager.cpp, function recvThreadCore(), line 340)
(Argus) Error InvalidState: Argus client is exiting with 1 outstanding client threads (in src/rpc/socket/client/ClientSocketManager.cpp, function recvThreadCore(), line 357)
(Argus) Error EndOfFile: Client thread received an error from socket (in src/rpc/socket/client/ClientSocketManager.cpp, function send(), line 145)
(Argus) Error EndOfFile:  (propagating from src/rpc/socket/client/SocketClientDispatch.cpp, function dispatch(), line 87)
(Argus) Error InvalidState: Receive thread is not running cannot send. (in src/rpc/socket/client/ClientSocketManager.cpp, function send(), line 96)
(Argus) Error InvalidState:  (propagating from src/rpc/socket/client/SocketClientDispatch.cpp, function dispatch(), line 87)
GST_ARGUS: Cleaning up
(Argus) Error InvalidState: Receive thread is not running cannot send. (in src/rpc/socket/client/ClientSocketManager.cpp, function send(), line 96)
(Argus) Error InvalidState:  (propagating from src/rpc/socket/client/SocketClientDispatch.cpp, function dispatch(), line 87)
Segmentation fault (core dumped)

isaac-223, 2021-03-23:
Hello @KevorkSulahian I am facing the same issue, did you solve it?

KevorkSulahian, 2021-03-23:
Hey @isaac-223  I unfortunately did not, if you do please make sure to write to me I will really appreciate it

init-22, 2021-03-25:
@KevorkSulahian so the error was related to camera which didn't have proper setup; I changed the camera and it worked!

carlinl, 2021-04-12:
I ran into the same issue and stuck for several days, I would be appreciated if someone can help. It shows the same issue on both CSI and USB camera.

KevorkSulahian, 2021-04-12:
Apparently you have to change the camera

carlinl, 2021-04-12:
Honestly, I don't think it's because of the camera, cuz both of them got the same error, and I tested USB camera on my MAC, it worked fine, also the type of the camera is compatible with Jetson Nano. Thanks anyway :P

KevorkSulahian, 2021-04-12:
Hope you're right and if you find a fix for it let me know because I can't
use mine

carlinl, 2021-04-13:
I solve this problem by reinstall the older version of the Jetson Nano image on SD-card. I tested the usb camera by installing camorama and cheese, after running cheese command it worked fine. I guess it might because there's some unsolved issue on latest version of image.

stephentandjiria, 2021-04-22:
Hi @carlinl I am also stuck on this problem with Jetpack 4.5.0.
Could you please tell me which older version of Jetpack image you installed?

carlinl, 2021-04-23:
I've installed Jetpack 4.4

stephentandjiria, 2021-04-23:
@carlinl thankyou, I really appreciate it!

Though for anyone reading this, at this point I still can't confirm Jetpack 4.4 works because my download for both 4.4 and 4.4.1 zip files are corrupted for unknown reasons, but I believe it's worth to try.

KavitShah1998, 2021-05-12:
> @KevorkSulahian so the error was related to camera which didn't have proper setup; I changed the camera and it worked!

I am facing the same issue with Jetson Nano, Jetpack 4.5.1 & raspi v2 camera.

@init-22 could you explain which camera were you using initially & what did you replace it with? Also what steps did you follow for the "setup" of your new camera?

I will also try out the method suggested by @carlinl, but wanted to see if there is any quick fix for it.

init-22, 2021-05-13:
@KavitShah1998 I was using Waveshare IMX219-160 and then I switched to  Arducom IMX291. Hope this helps

deeprun, 2021-08-20:
@init-22 I am using waveshare IMX219-160 and it worked in the past.Now it did not work. I am waiting for my Arducom IMX291.

radhen, 2022-01-07:
> ave to change the camera

I have tried three cameras, and yet get the same error. Did anyone find a solution to this problem?

radhen, 2022-01-07:
> I solve this problem by reinstall the older version of the Jetson Nano image on SD-card. I tested the usb camera by installing camorama and cheese, after running cheese command it worked fine. I guess it might because there's some unsolved issue on latest version of image.

 @carlinl , which JetPack version work for you?

Moddingear, 2022-04-27:
I have the same issue, and it doesn't seem to be a camera issue : using `gst-launch-1.0 nvarguscamerasrc ! nvegltransform ! nveglglessink` works, but using the `nvvidconv ! videoconvert` pipeline doesn't.
This seems to be a configuration issue somewhere.
Furthermore, the `nvvidconv ! videoconvert` pipeline worked before I used it in opencv, where it only worked once...
Has anyone gotten more insights into this issue since the last post ?

Moddingear, 2022-10-25:
> > I have the same issue, and it doesn't seem to be a camera issue : using `gst-launch-1.0 nvarguscamerasrc ! nvegltransform ! nveglglessink` works, but using the `nvvidconv ! videoconvert` pipeline doesn't. This seems to be a configuration issue somewhere. Furthermore, the `nvvidconv ! videoconvert` pipeline worked before I used it in opencv, where it only worked once... Has anyone gotten more insights into this issue since the last post ?
> 
> Could you please share your full gsteamer pipelines? Both with `nvvidconv ! videoconvert` and with `gst-launch-1.0 nvarguscamerasrc ! nvegltransform ! nveglglessink`.

In the end I got opencv working : I was using a library that listed v4l2 devices that interacted incorrectly with the drivers for nvargus. Updating the library or removing it fixed the issue.

UUUUUUsername, 2022-11-27:
Hi, @Moddingear , I have met the save problem. Could you tell me what library did you update or remove?

Moddingear, 2022-11-27:
> Hi, @Moddingear , I have met the save problem. Could you tell me what library did you update or remove?

It was this library : https://github.com/doleron/v4l2-list-devices
The 9 June 2022 version works, the earlier ones don't.

UUUUUUsername, 2022-11-27:
> > Hi, @Moddingear , I have met the save problem. Could you tell me what library did you update or remove?
> 
> It was this library : https://github.com/doleron/v4l2-list-devices The 9 June 2022 version works, the earlier ones don't.

Thank you for your reply! I just figured that the problem is that I did not add -D WITH_GSTREAMER when installing opencv.