JetsonHacksNano/CSI-Camera #38: [BUG] Unable to open camera when building opencv from source

squeakus, 2021-04-19:
**Describe the issue**
Hi,I am unable to access the camera when I build from source, even with gstreamer enabled. I get the following error

nvarguscamerasrc ! video/x-raw(memory:NVMM), width=(int)1280, height=(int)720, format=(string)NV12, framerate=(fraction)60/1 ! nvvidconv flip-method=0 ! video/x-raw, width=(int)1280, height=(int)720, format=(string)BGRx ! videoconvert ! video/x-raw, format=(string)BGR ! appsink
Unable to open camera

If I uninstall and install from apt it works fine so it must be some other dependency I am missing. I need to build it from source to enable the nvidia optical flow contrib packages

**What version of L4T/JetPack**
L4T/JetPack version:
R32 (release), REVISION: 5.1, GCID: 26202423, BOARD: t210ref, EABI: aarch64, DATE: Fri Feb 19 16:45:52 UTC 2021
**What version of OpenCV**
OpenCV version:
4.5
**Python Version**
3.8

**To Reproduce**
Steps to reproduce the behavior:
For example, what command line did you run?
python3 simple_camera.py

**Expected behavior**
run the camera app and show output

**Additional context**
Add any other context about the problem here.

squeakus, 2021-04-19:
My bad, when I checked the build with print(cv2.getBuildInformation()) it turned out that gstreamer was failing silently I needed to install additional gstreamer dev packages.

sudo apt install libgstreamer1.0-dev libgstreamer-plugins-base1.0-dev

closed 2021-04-19 by squeakus

Rusali28, 2023-03-20:
Hey, I am facing a similar issue. I tried installing the additional packages, but I am still getting the same error, can you please give me any idea how to solve it?