JetsonHacksNano/CSI-Camera #42: [BUG] Throw segmentation fault when execute the simple_camera.py

jszhtian, 2021-05-16:
**Describe the issue**
The simple_camera.py throw segmentation fault when press the "ESC" key to exit the application.

**What version of L4T/JetPack**
L4T/JetPack version: Jetpack 4.5.1 [L4T 32.5.1]

**What version of OpenCV**
OpenCV version: OpenCV: 4.1.1 compiled CUDA: NO

**Python Version**
Python version if applicable: Python 3.6.9

**To Reproduce**
Direct execute python3 ./simple_camera.py
Wait few seconds (larger than 15 seconds)
Press "ESC" to exit the application

**Expected behavior**
Exit the application without errors.

**Additional context**
Add any other context about the problem here.
HW&SW:
 - NVIDIA Jetson Nano (Developer Kit Version)
   * Jetpack 4.5.1 [L4T 32.5.1]
   * jetson_stats.service: active
 - Libraries:
   * CUDA: 10.2.89
   * cuDNN: 8.0.0.180
   * TensorRT: 7.1.3.0
   * Visionworks: 1.6.0.501
   * OpenCV: 4.1.1 compiled CUDA: NO
   * VPI: ii libnvvpi1 1.0.15 arm64 NVIDIA Vision Programming Interface library
   * Vulkan: 1.2.70

[image]

S4WRXTTCS, 2021-05-22:
I'm getting the same issue. 

In my case I'm using a newer Jetson Nano development board (2 camera ports) along with the Jetpack 4.5.1. For the camera I'm using an Arducam IMX477.

One interesting thing was I tried using an older Jetson Nano development board with the same Jetpack, but it gives me an unable to open camera error message using Python3, but is able to open the camera using Python 2.7. I'm not too worried about that one as the segmentation fault prevents using it cleanly.

jszhtian, 2021-05-22:
> I'm getting the same issue.
>
> In my case I'm using a newer Jetson Nano development board (2 camera ports) along with the Jetpack 4.5.1. For the camera I'm using an Arducam IMX477.
>
> One interesting [...]

I found a not perfect solution for this problem. Just rollback to jetpack 4.4.1 or older version, no errors will be throw. Maybe it's Nvidia's problem. Actually, the newer jetpack version add new platforms support, and almost no features update for Jetson nano.