JetsonHacksNano/CSI-Camera #44: There is a problem that it cannot run in python3.

Choikyungho9, 2021-05-31:
simple_test.py works fine with python simple_test.py on jetpack 4.4.
But python3 simple_test.py doesn't work.

It says Unable to open camera.

Please let me know what the problem is.

I'm using python 3.6.9 version / opencv 4.5.1.

jksojitraa, 2022-03-01:
@Choikyungho9
@jetsonhacks 
hello there,

Have you solved your problem or not please let me know if you solved it.
it will be nice if you explain how you solved it.

thanks in advance

> peace

JetsonHacksNano, 2022-03-01:
Which version are you running? I've been unable to recreate this.

jksojitraa, 2022-03-01:
thanks for replying but I have solved it.

there are issues with OpenCV.
in python3 I have installed the OpenCV - 4.4.5
and in python2 I have been using the pre-installed OpenCV - 4.1.1 in jetson nano os 
so it is working well with python2 but can not able to open the pipeline by using OpenCV-4.4.5.

So I was checking the OpenCV versions and found the solution.
then just simply removed the OpenCV-4.5.5 from python3. and now working very well.

I have never doubted the @jetsonhacks codes It always runs in like one attempt. no west of time. my bad🤗

--
--
thanks and regards 
jay sojitra