JetsonHacksNano/CSI-Camera #51: It doesn't work when I've configured it[BUG]

yinqiyu, 2022-02-19:
**Describe the issue**
After configuring the Gstream,I want to make calls to the CSI camera through the darknet framework, it is stuck in the picture shown in the picture, and then the jetson nano will restart.
[image]

**What version of L4T/JetPack**
L4T/JetPack version:4.4

**What version of OpenCV**
OpenCV version:opencv4.1.1

**Python Version**
Python version if applicable:python3.6

**To Reproduce**
Steps to reproduce the behavior:
./darknet detector demo cfg/coco.data cfg/yolov4-tiny.cfg yolov4-tiny.weights "nvarguscamerasrc ! video/x-raw(memory:NVMM), width=1280, height=720, format=NV12, framerate=30/1 ! nvvidconv flip-method=2 ! video/x-raw, width=1280, height=720, format=BGRx ! videoconvert ! video/x-raw, format=BGR ! appsink"

JetsonHacksNano, 2022-02-19:
This does not have anything to do with the code in this repository. Please ask this question on the official NVIDIA Jetson forums: https://forums.developer.nvidia.com/c/agx-autonomous-machines/jetson-embedded-systems/70

yin-qiyu, 2022-02-19:
ok tanks so much

closed 2022-02-19 by yin-qiyu