JetsonHacksNano/CSI-Camera #52: Unable to open camera when running simple_camera.py with Python version 3.7.5

daniellm19, 2022-03-22:
**Describe the issue**
Unable to open camera when running simple_camera.py with Python version 3.7.5 using VSCode.

**What version of L4T/JetPack**
L4T/JetPack version: 4.6.1

**What version of OpenCV**
OpenCV version: 4.1.1.2 (although some seem to be 3.2.0 when I use the command: "dpkg -l | grep libopencv")

**Python Version**
Python version if applicable: 3.7.5

**To Reproduce**
Steps to reproduce the behavior:
1. Power the Jetson Nano
2. Attach RPi V2 camera
3. Clone this repo
4. Run the simple_camera.py file

**Expected behavior**
A Gstreamer window pops up which shows the camera working, like it does for python 2.7.17

**Additional context**
1. I have a ton of different installations and overall, it's kind of a mess. Locating the root issue might be rather problematic.
2. I have used a command that removes python from the computer so that I could install the 3.7 version, what issues might this have caused? Well I needed to at least reinstall pip, who knows what else it did.

AINacster, 2023-04-02:
Can anyone help me with this issues. Simple_camera.py works with 2.7 but when I run it python3 (3.8) it errors out saying not able to open camera. 

Appreciate if you can help me with this issues. This is on Nvidia Jetson Nano.

Kerem-Kurt, 2023-08-16:
I am using python3.8 and I am getting
ImportError: cannot import name '_gi' from partially initialized module 'gi' (most likely due to circular import) (/usr/lib/python3/dist-packages/gi/__init__.py)