JetsonHacksNano/CSI-Camera #55: unable to open camera [BUG]

Daryatnt, 2022-04-15:
I'm facing this error while I'm trying to compile the code :
```
(python3:287093): GStreamer-WARNING **: 19:10:20.532: Failed to load plugin '/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstnvarguscamerasrc.so': /usr/lib/aarch64-linux-gnu/libGLdispatch.so.0: cannot allocate memory in static TLS block

(python3:287093): GStreamer-WARNING **: 19:10:20.534: Failed to load plugin '/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstnvvidconv.so': /usr/lib/aarch64-linux-gnu/libGLdispatch.so.0: cannot allocate memory in static TLS block
[ WARN:0] global /home/jetson/opencv/modules/videoio/src/cap_gstreamer.cpp (734) open OpenCV | GStreamer warning: Error opening bin: no element "nvarguscamerasrc"
[ WARN:0] global /home/jetson/opencv/modules/videoio/src/cap_gstreamer.cpp (501) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
[ERROR:0] global /home/jetson/opencv/modules/videoio/src/cap.cpp (142) open VIDEOIO(CV_IMAGES): raised OpenCV exception:

OpenCV(4.5.1) /home/jetson/opencv/modules/videoio/src/cap_images.cpp:253: error: (-5:Bad argument) CAP_IMAGES: can't find starting number (in the name of file): nvarguscamerasrc !  video/x-raw(memory:NVMM), width=1640, height=1232, format=NV12, framerate=21/1 ! nvvidconv flip-method=0 ! video/x-raw, width=960, height=720, format=BGRx !videoconvert ! video/x-raw, format=BGR ! appsink in function 'icvExtractPattern'
```

**What version of L4T/JetPack**
4.6

**What version of OpenCV**
4.5.1

**Python Version**
3.8

YavuzKacar, 2022-05-16:
I am having the same problem, any progress about the solution? :)

YavuzKacar, 2022-05-18:
Uninstalling opencv-python and installing it with the following command fixed my problem:
`pip3 install git+https://github.com/opencv/opencv-python`

dleporis, 2022-11-28:
Try an OpenCV version specified in the README.md release notes. Keep in mind 4.4.1 JetsonHacksNano wrote was a typo. But 3.3.1 and 4.1.1 worked for me.