JetsonHacksNano/CSI-Camera #62: [BUG] Newly set up Jetson Orin Nano has no nvarguscamerasrc or nvoverlaysink

JetForMe, 2023-04-18:
**Describe the issue**

I was going through the README to get my two RPi cameras up and running. The Python examples work just fine (e.g. `python3 dual_camera.py` displays the output of the two connected cameras). But these commands fail:

```
$ nvoverlaysink
-bash: nvoverlaysink: command not found
$ gst-launch-1.0 nvarguscamerasrc sensor_id=0 ! nvoverlaysink
WARNING: erroneous pipeline: no element "nvoverlaysink"
```

Clearly something isn't installed, but my googling hasn't turned up an answer.

**What version of L4T/JetPack**
JetPack 5.1.1/L4T35.3.1

**What version of OpenCV**
OpenCV version: 4.5.4

**Python Version**
Python version if applicable: 3.8.10

**To Reproduce**

See description above

**Expected behavior**

No errors about missing commands or plug-ins.

JetForMe, 2023-04-18:
Ah, it seems those are deprecated (according to the Jetson forums).

JetForMe, 2023-05-31:
I had to put the project aside for a while, so I forgot how far I was able to get, but check this [answer](https://forums.developer.nvidia.com/t/nvarguscamerasrc-nvoverlaysink-missing/250307/3)

skydrive22, 2025-02-11:
I just started my project and found it runs if you replace nvoverlaysink with nveglglessink. I'm using a csi camera and I needed to reconfigure the csi port for imx219-A

gst-launch-1.0 nvarguscamerasrc ! nvvidconv ! nveglglessink

AaronSif, 2025-02-15:
If i try gst-launch-1.0 nvarguscamerasrc sensor-id=0 ! nvdrmvideosink) it looks like everything runs find but the camera doesn't actually launch?  Below is the result.

nano1@nano1-desktop:~$ gst-launch-1.0 nvarguscamerasrc sensor-id=0 ! nvdrmvideosink
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
GST_ARGUS: Creating output stream
CONSUMER: Waiting until producer is connected...
GST_ARGUS: Available Sensor modes :
GST_ARGUS: 3280 x 2464 FR = 21.000000 fps Duration = 47619048 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 3280 x 1848 FR = 28.000001 fps Duration = 35714284 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1920 x 1080 FR = 29.999999 fps Duration = 33333334 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1640 x 1232 FR = 29.999999 fps Duration = 33333334 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1280 x 720 FR = 59.999999 fps Duration = 16666667 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: Running with following settings:
   Camera index = 0 
   Camera mode  = 2 
   Output Stream W = 1920 H = 1080 
   seconds to Run    = 0 
   Frame Rate = 29.999999 
GST_ARGUS: Setup Complete, Starting captures for 0 seconds
GST_ARGUS: Starting repeat capture requests.
CONSUMER: Producer has connected; continuing.
ERROR: from element /GstPipeline:pipeline0/GstNvArgusCameraSrc:nvarguscamerasrc0: Internal data stream error.
Additional debug info:
../libs/gst/base/gstbasesrc.c(3127): gst_base_src_loop (): /GstPipeline:pipeline0/GstNvArgusCameraSrc:nvarguscamerasrc0:
streaming stopped, reason not-negotiated (-4)
Execution ended after 0:00:00.394113481
Setting pipeline to NULL ...
GST_ARGUS: Cleaning up
CONSUMER: Done Success
GST_ARGUS: Done Success
Freeing pipeline ...

skydrive22, 2025-02-22:
Interesting, are you using a csi camera? Have you tried specifying a specific format? 

gst-launch-1.0 nvarguscamerasrc sensor-id=0 ! 'video/x-raw(memory:NVMM), width=1920, height=1080, format=NV12, framerate=30/1' ! nvdrmvideosink

Try running this using NV12 format and see if that works

AaronSif, 2025-02-23:
@skydrive22 

Thanks for the response!  Yes I am using a CSI camera.  If i use the script you wrote above Terminal still seems as though everything indexes through but then the camera doesn't actually load?  Below is the output from terminal if i copy and paste the script you wrote above.  

I should note that i am able to get the camera to open using some other scripts that use "gst-launch", such as "gst-launch-1.0 nvarguscamerasrc ! 'video/x-raw(memory:NVMM),framerate=30/1,format=NV12' ! nvvidconv ! xvimagesink" but not using the above commands from the tutorial.

Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
GST_ARGUS: Creating output stream
CONSUMER: Waiting until producer is connected...
GST_ARGUS: Available Sensor modes :
GST_ARGUS: 3280 x 2464 FR = 21.000000 fps Duration = 47619048 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 3280 x 1848 FR = 28.000001 fps Duration = 35714284 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1920 x 1080 FR = 29.999999 fps Duration = 33333334 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1640 x 1232 FR = 29.999999 fps Duration = 33333334 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1280 x 720 FR = 59.999999 fps Duration = 16666667 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: Running with following settings:
   Camera index = 0 
   Camera mode  = 2 
   Output Stream W = 1920 H = 1080 
   seconds to Run    = 0 
   Frame Rate = 29.999999 
GST_ARGUS: Setup Complete, Starting captures for 0 seconds
GST_ARGUS: Starting repeat capture requests.
CONSUMER: Producer has connected; continuing.
ERROR: from element /GstPipeline:pipeline0/GstNvArgusCameraSrc:nvarguscamerasrc0: Internal data stream error.
Additional debug info:
../libs/gst/base/gstbasesrc.c(3127): gst_base_src_loop (): /GstPipeline:pipeline0/GstNvArgusCameraSrc:nvarguscamerasrc0:
streaming stopped, reason not-negotiated (-4)
Execution ended after 0:00:00.412528122
Setting pipeline to NULL ...
GST_ARGUS: Cleaning up
CONSUMER: Done Success
GST_ARGUS: Done Success
Freeing pipeline ...